SlimeVR/SlimeVR-Tracker-nRF #24: NVS errors

42triangles, 2025-01-13:
See https://discord.com/channels/817184208525983775/1209951319209680946/1319634604138958861

Supposedly
```dto
&flash0 {
    partitions {
        compatible = "fixed-partitions";
        #address-cells = <1>;
        #size-cells = <1>;

        boot_partition: partition@0 {
            label = "mbr";
            reg = <0x0 0x1000>;
        };
        slot0_partition: partition@1000 {
            label = "image-0";
            reg = <0x1000 0x76000>;
        };
        slot1_partition: partition@77000 {
            label = "image-1";
            reg = <0x77000 0x76000>;
        };
        storage_partition: partition@ed000 {
            label = "storage";
            reg = <0xed000 0x6000>;
        };
        boot_partition1: partition@f4000 {
            label = "bootloader";
            reg = <0xf4000 0xc000>;
        };
    };
};
```
gets rid of the NVS errors.

Original message by toaste_:
>  Messages regarding NVS are gone when using the map below
> (the app partitions are still made up, but I would guess this doesn't work well for mcuboot)
> You will note this map wastes a 4k gap betwen storage and the bootloader. This is because I have trust issues.

sctanf, 2025-01-14:
I wonder if it is a hardware fault, and the issue only occurred on some board or on all of the board type? Partition Manager will be replacing manual partitions, this should hopefully also resolve the issue.

sctanf, 2025-02-28:
partition manager setup should now avoid bootloader partitions with nvs

sctanf, 2025-04-09:
closing this since fix implemented and no feedback, reopen if any issues

closed 2025-04-09 by sctanf