SlimeVR/SlimeVR-Tracker-nRF #30: Feature Request: Automatic Deep Sleep

ManicQuinn, 2025-02-18:
A opt-in feature in firmware for the trackers to automatically go into deep sleep after a adjustable amount of time(an hour, for example).
Assume it might be easy to implement since both sleep and no sleep are already tracking inactive trackers. Would prefer opt in for those who would prefer the trackers running their whole cycle, say for sleeping in VR for example.

sctanf, 2025-02-18:
assuming they can be undisturbed for that long, sounds good!
i think the logic can be changed a little, by letting the tracker wake up normally, which might be better if you use the feature to sleep with trackers on
it could still enter deep sleep if you use the feature for after taking trackers off, or wanted to use it to sleep in VR anyway

sctanf, 2025-02-26:
implemented as activity timeout and can be configured by kconfig, the default is IMU wake up

closed 2025-03-10 by ManicQuinn