SlimeVR/SlimeVR-Tracker-nRF #93: Pre-Compiled Firmware issue

bitdeckai, 2025-07-12:
Hi, 

I have a nRF52840 XIAO sense board with IMU sensor LSM6D3TR-C, I haved download the bootloader, then which uf2 file **Clock (ICM)** or **Sleep (WOM)** or **SW0 Disabled/Enabled** ?

another problem, I copy the file , but don't detect IMU, why ?

[image]

ShineBrightMeow, 2025-07-12:
The pre-compiled firmware listed on the docs are for Seeed Studio XIAO nRF52840 (non-Sense) version to be used with an external IMU.

Try these for Sense:
https://github.com/Shine-Bright-Meow/SlimeNRF-Firmware-CI/releases/download/latest/SlimeNRF_Tracker_XIAO_Sense.uf2
https://github.com/Shine-Bright-Meow/SlimeNRF-Firmware-CI/releases/download/latest/SlimeNRF_Tracker_SW0_XIAO_Sense.uf2
https://github.com/Shine-Bright-Meow/SlimeNRF-Firmware-CI/releases/download/latest/SlimeNRF_Tracker_NoSleep_XIAO_Sense.uf2
https://github.com/Shine-Bright-Meow/SlimeNRF-Firmware-CI/releases/download/latest/SlimeNRF_Tracker_SW0_NoSleep_XIAO_Sense.uf2

bitdeckai, 2025-07-12:
Hi, ShineBrightMeow:

Thanks, I haved scan the IMU, and how to connect to SlimeVR server ?
（1）XIAO nRF52840 (non-Sense) as Receiver
（2）XIAO nRF52840 Sense as Tracker

[image]

bitdeckai, 2025-07-12:
Hi

Reboot the all device, SlimeVR server have display IMU data, 
 Thank you very much!

[image]

closed 2025-07-12 by ShineBrightMeow