SlimeVR/SlimeVR-Tracker-nRF #98: how to add build configuration based on promicro board?

bitdeckai, 2025-07-18:
Hi, everyone:

The sensor is ICM-45686 and QMC6309 based on ProMicro Board. How to add build configuration? 

[image]

ShineBrightMeow, 2025-07-18:
It looks fine. What seems to be the issue? Are you not using the default pins?
There are pre-compiled firmware versions available in the Docs.

If you're creating a new variant, you'll need to create a new partition (`pm`) file, additional board files (`.dts`, `defconfig`, and `.yml`), and add the variant to `board.yml`.

ShineBrightMeow, 2025-07-18:
That's just the build configuration—the firmware doesn't require you to specify the IMU or magnetometer.

If you're using the default pins, why not just use the pre-compiled firmware from here?
https://docs.slimevr.dev/smol-slimes/firmware/smol-pre-compiled-firmware.html

ShineBrightMeow, 2025-07-18:
Alternatively, you can build it using GitHub Actions, but you'll need to fork the repository first. See `.github\workflows\workflow.yml`.

closed 2025-07-19 by bitdeckai