SlimeVR/SlimeVR-Tracker-nRF #102: Suggesting a workaround for a hardware oopsie on ProMicro boards

kkpdk, 2025-08-01:
I was debugging a few trackers where the ProMicro NRF52840 can't talk to the IMU when on battery; It is voltage dependent, on USB or 5V it works.

The ME6217 3.3V regulator is enabled by (as far as I can trace) P0.13, which has a 10MOhm pullup to input voltage.
On these boards the enable signal hovers around 0.5V; While in theory the regulator shouldn't have more than 100nA leakage at room temperature, there is also the leakage in the pin of the uC, and as a result the 200nA the pull-up can provide (to get >=1.5V on enable) is not enough.
The high impedance, and the not-in-spec enable voltage, even on 'good' boards, makes most of them have problems on USB plug/unplug.

Putting on my hardware engineer's hat, this is a hardware bug.
But a software workaround would be to drive enable high early in the program, and allow enough time for the IMU to go through POR before talking to it.

/kkpdk

(Pin 3 of the ME6217 is enable.
I used an HP34401A in Hi-Z mode for the measurement on enable, snd desoldered the resistor to measure it)

sctanf, 2025-08-02:
Thanks for the insight! However, since the current bootloader compiled for the ProMicro sets regout0 to 2.1V, it is now recommended to use P0.31 to power the sensor instead of 3V3.

kkpdk, 2025-08-02:
I'd say 'mandatory'; the output of the IMU in SPI mode will be 3.3V when supplied from 3.3V, so it's violating absolute maximum ratings on the input, and, as I read the datasheet, raising VDD through the pin protection diode. But this choice makes sense.

8 trackers modified, flashed with the current bootloader, voltage and current checked, and behaving well at first glance (Bricked one, have jlink.)

sctanf, 2025-08-11:
At request of @kounocom and @brisfknibis, P0.13 is now pulled up by default, which should be enough to keep the pin high. While sensors should not be connected to 3V3, a board design may still need the regulator to function (such as status lights).

closed 2025-08-11 by sctanf